andy-shev/linux #2: eds branch kernel works in 32-bit mode, but not in 64-bit mode

cklin, 2016-09-14:
Is the kernel from eds branch supposed to work on Edison in 64-bit mode?

If I build the kernel (current eds branch head, 01e5c69a4f4a13d43e753157a5602b1d8c690469) in i386 architecture, the kernel boots fine on Intel Edison with mini-breakout board.  But if I build with x86_64 architecture using the same kernel source, the kernel fails to boot on the same Edison board.  This is what I see on the serial console:

```
reading vmlinuz2
6727072 bytes read in 164 ms (39.1 MiB/s)
Valid Boot Flag
Setup Size = 0x00003e00
Magic signature found
Using boot protocol version 2.0d
Linux kernel version 4.8.0-rc6-barb+ (user@common) #1 SMP Tue Sep 13 21:29:38 PDT 2016
Building boot_params at 0x00090000
Loading bzImage at address 00100000 (6711200 bytes)
Magic signature found
Kernel command line: "rootwait root=PARTUUID=012b3303-34ac-284d-99b4-34e03a2335f4 rootfstype=ext4 console=tty1 console=ttyS2,115200n8 console=ttyMFD2 loglevel=8 g_multi.ethernet_config=cdc systemd.unit=multi-user.target hardware_id=00 g_multi.iSerialNumber=28b8e149bda6b46f643f266147d3be59 g_multi.dev_addr=02:00:86:d3:be:59 platform_mrfld_audio.audio_codec=dummy"

Starting kernel ...
```

And then nothing happens, until the watchdog kicks in and reboots the system.

I built the kernels using the standard "make -j4" command, with gcc (Debian 6.1.1-11) 6.1.1 20160802 on a x86_64 Linux system running Debian testing.  The kernel configs are generated from i386_defconfig and x86_64_defconfig, respectively, and everything else (even the kernel boot arguments provided by u-boot) are the same.

The page https://edison.internet-share.com/wiki/Using_a_vanilla_Linux_kernel_with_Intel_Edison states that Linux 4.4 boots on Edison in 64-bit mode, so I was hoping that the latest eds branch would also work in 64-bit mode.  But perhaps there has been a regression?

Any help would be appreciated.  Please let me know if I can provide more information to help with troubleshooting.  Thanks!

[config-x86_64.txt](https://github.com/andy-shev/linux/files/471296/config-x86_64.txt)

andy-shev, 2016-09-23:
You have to update U-Boot to at least 2015.05. I suppose the one from here should work (never tested): https://github.com/01org/edison-u-boot/tree/edison-v2016.03 (you may try starting from older one).

cklin, 2016-09-24:
I see.  Thanks for the pointer - though I am a bit reluctant to mess with custom u-boot.  I am confident that I can recover from a bad Linux kernel, but a bad custom u-boot image seems like the kind of problem that can brick an Edison board.

andy-shev, 2016-11-18:
Here is the recipe. https://communities.intel.com/message/435516#435516

closed 2016-11-18 by andy-shev