andy-shev/linux #4: Wi-fi support

zyp, 2016-11-14:
Hi, I'm trying to get wi-fi to work, but I'm having no luck so far. I learned that mainline 4.8 kernel doesn't have the patches to enable the mmc host and regulator for the wi-fi chip, but even the current tip (2cbbee3) of your eds branch doesn't seem to work.

Excerpts from dmesg:
```
[    0.129318] bcm43xx-vmmc-regulator: Failed to request enable GPIO96: -517
[    0.129541] reg-fixed-voltage reg-fixed-voltage.0.auto: Failed to register regulator: -517
…
[    0.705322] sdhci: Secure Digital Host Controller Interface driver
[    0.705327] sdhci: Copyright(c) Pierre Ossman
[    0.705394] sdhci-pci 0000:00:01.0: SDHCI controller found [8086:1190] (rev 1)
[    0.714112] mmc0: SDHCI controller on PCI [0000:00:01.0] using ADMA
[    0.714203] sdhci-pci 0000:00:01.2: SDHCI controller found [8086:1190] (rev 1)
[    0.723140] mmc1: SDHCI controller on PCI [0000:00:01.2] using ADMA
[    0.723417] sdhci-pci 0000:00:01.3: SDHCI controller found [8086:1190] (rev 1)
[    0.732260] mmc2: SDHCI controller on PCI [0000:00:01.3] using ADMA
[    0.732511] sdhci-pltfm: SDHCI platform and OF driver helper
…
[    0.752916] bcm43xx-vmmc-regulator: at 2000 mV 
[    0.756369] reg-fixed-voltage reg-fixed-voltage.0.auto: bcm43xx-vmmc-regulator supplying 2000000uV
…
[    0.756812] bcm43xx-vmmc-regulator: disabling
```

Modprobing brcmfmac does not result in any output to dmesg.

The first line looks a bit suspect, but I assume it's because the GPIO driver is not initialized yet at the time. It seems like the regulator driver gets reinitialized at some point because `/sys/kernel/debug/gpio` later indicates it's controlling the GPIO.

```
gpiochip0: GPIOs 0-191, parent: pci/0000:00:0c.0, 0000:00:0c.0:
 gpio-77  (                    |sd_cd               ) in  hi IRQ
 gpio-96  (                    |bcm43xx-vmmc-regulat) out lo    
```

I tried removing the regulator driver and controlling GPIO 96 manually, but that didn't help.

andy-shev, 2016-11-16:
Everything works, though there is one limitation: one may not have **sdhci** and **brcmfmac** compiled in the kernel. Try the config I supply with my branch w/o any change. 
P.S. In case you need to boot from SD or need it be available earlier, feel free to create an *initrd* where you load necessary modules and use `pivot_root` to switch /dev/root to a correct one.

closed 2016-11-16 by andy-shev

andy-shev, 2016-12-07:
@ecc1, I have got an Intel Mini-Breakout Board and tried my kernel on it. I see Wi-Fi module is loaded and no other issues. Yes, I have tried x86_64 version for now. Please, tell me if you are using 32-bit one.

andy-shev, 2016-12-07:
Just from i686 version from today's eds branch (nothing new except maybe v4.9-rc8):
<pre>
# dmesg | grep -E 'regulator|mmc|sdhci|sdio|brcm|bcm'
[    0.079508] regulator-dummy: no parameters
[    0.128468] bcm43xx-vmmc-regulator: Failed to request enable GPIO96: -517
[    0.128721] reg-fixed-voltage reg-fixed-voltage.0.auto: Failed to register regulator: -517
[   14.988830] bcm43xx-vmmc-regulator: at 2000 mV 
[   14.989052] reg-fixed-voltage reg-fixed-voltage.0.auto: bcm43xx-vmmc-regulator supplying 2000000uV
[   15.001217] bcm43xx-vmmc-regulator: disabling
[   16.075861] sdhci: Secure Digital Host Controller Interface driver
[   16.082124] sdhci: Copyright(c) Pierre Ossman
[   16.100662] sdhci-pci 0000:00:01.0: SDHCI controller found [8086:1190] (rev 1)
[   16.118022] mmc0: SDHCI controller on PCI [0000:00:01.0] using ADMA
[   16.124705] sdhci-pci 0000:00:01.2: SDHCI controller found [8086:1190] (rev 1)
[   16.146098] mmc1: SDHCI controller on PCI [0000:00:01.2] using ADMA
[   16.153515] sdhci-pci 0000:00:01.3: SDHCI controller found [8086:1190] (rev 1)
[   16.173096] mmc2: SDHCI controller on PCI [0000:00:01.3] using ADMA
[   16.271028] mmc0: new DDR MMC card at address 0001
[   16.312525] mmcblk0: mmc0:0001 H4G1d 3.64 GiB 
[   16.317552] mmcblk0boot0: mmc0:0001 H4G1d partition 1 4.00 MiB
[   16.324475] mmcblk0boot1: mmc0:0001 H4G1d partition 2 4.00 MiB
[   16.330875] mmcblk0rpmb: mmc0:0001 H4G1d partition 3 4.00 MiB
[   16.357216]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10
[   16.543972] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[   16.552100] mmc2: queuing unknown CIS tuple 0x80 (6 bytes)
[   16.559128] mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
[   16.566744] mmc2: queuing unknown CIS tuple 0x80 (4 bytes)
[   16.665979] mmc2: new ultra high speed DDR50 SDIO card at address 0001
[   17.064440] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[   17.072890] mmc2: queuing unknown CIS tuple 0x80 (6 bytes)
[   17.079862] mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
[   17.087487] mmc2: queuing unknown CIS tuple 0x80 (4 bytes)
[   17.148477] brcmfmac: F1 signature read @0x18000000=0x1602a94c
[   17.158601] brcmfmac mmc2:0001:1: Direct firmware load for brcm/brcmfmac43340-sdio.txt failed with error -2
[   18.175636] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[   19.191462] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
# uname -a
Linux buildroot 4.9.0-rc8+ #2 SMP Wed Dec 7 17:13:21 EET 2016 i686 GNU/Linux
</pre>

andy-shev, 2016-12-07:
Just notice that *Direct firmware load for brcm/brcmfmac43340-sdio.txt failed* message means one needs to copy calibration file with given name.

andy-shev, 2016-12-07:
After calibration file is copied and module is reloaded I have got the following:
<pre>
[  602.360443] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[  602.368764] mmc2: queuing unknown CIS tuple 0x80 (6 bytes)
[  602.375799] mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
[  602.383436] mmc2: queuing unknown CIS tuple 0x80 (4 bytes)
[  602.704640] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Jun 17 2014 11:48:43 version 6.10.190.49 (r485840) FWID 01-8744bcd2
[  602.742882] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code (0x30 0x30)

# ip link list
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: sit0@NONE: <NOARP> mtu 1480 qdisc noop qlen 1
    link/sit 0.0.0.0 brd 0.0.0.0
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether fc:c2:de:45:2a:82 brd ff:ff:ff:ff:ff:ff
</pre>

zyp, 2016-12-07:
I can confirm that I got wi-fi working after changing sdhci to be built as a module.

It's a slight hassle to have to deal with an initrd though, so it would be nice to fix it so it works even when sdhci is built into the kernel. You're mentioning boot from SD, but sdhci needs to be loaded to access the EMMC too.

ecc1, 2016-12-07:
Having sdhci compiled-in has probably been my problem too.  I'm trying to drop in a kernel in a system that boots from MMC, currently with no initrd, so I'll have to change the boot sequence a bit to test.

andy-shev, 2016-12-08:
@zyp, yeah, I know, you need to create a *"trampoline" initrd* which loads modules and switches to an actual rootfs. So many distros are doing this way. That's why I consider this issue as low priority and severity.

andy-shev, 2017-01-03:
Just for reference [the discussion about regulator framework issue](http://www.spinics.net/lists/kernel/msg2333536.html) in case of Edison. It would be fixed either by switching to ACPI, or by patching the framework if anyone provides a such. I would prefer a), i.e. switching to ACPI, and thus will not go to spend my time to fix the regulator framework code, though I could help anyone who will try b).

htot, 2017-05-25:
andy above you mention that you need to copy the calibration file brcm/brcmfmac43340-sdio.txt. Where do you copy that from?

andy-shev, 2017-05-26:
You may take it from Yocto, or from existing stock image (*/etc/firmware/* as far as I remember).

http://git.yoctoproject.org/cgit/cgit.cgi/meta-intel-edison/tree/meta-intel-edison-bsp/recipes-kernel/bcm43340/bcm43340-fw.bb

htot, 2017-05-26:
That recipe installs 2 cal files (bcmdhd_aob.cal and bcmdhd.cal) and 1 bin. So I guess it is one of those cal files and it needs to be renamed?

htot, 2017-05-31:
I think I understand:
bcmdhd.cal is for country code ALL
bcmdhd_aob.cal for FR, it think it disables 5G, not sure. 
According to this doc http://download.intel.com/support/edison/sb/edison_wifi_331438001.pdf should not be present, maybe it's obsolete.

With both the firmware (bin) and bcmdhd.cal file renamed, wifi works nicely on both 2.4G and 5G. This of course when booting using initramfs with mmc and sdhci modules included and then using switch_root to trampoline the rootfs.