andy-shev/linux #5: IRQ disabling on Edison 64-bit kernel

amcduffee, 2016-11-22:
I have built a 64-bit 4.9 kernel from your 'eds' (8624493) branch and shortly after booting all the way to a command prompt I get an IRQ disabled stack trace.  

[dmesg.txt](https://github.com/andy-shev/linux/files/607722/dmesg.txt)
[edison64_config.txt](https://github.com/andy-shev/linux/files/607723/edison64_config.txt)
[proc_interrupts.txt](https://github.com/andy-shev/linux/files/607724/proc_interrupts.txt)

Let me know what other details I should gather.

andy-shev, 2016-11-23:
Yeah, I know about the issue. If you are okay not to use DMA for UART, you may try to avoid that. Requires a bit of a hacking code (_8250_mid.c_: comment out `dma_dev=` in `tng_setup()` function).

VladimirAkopyan, 2016-12-08:
Is making a 64 bit kernel relatively straight forward? I've been using Edison for a while, but never had to compile a kernel. We happen to need a 64 bit version. 
About this - we do use UART, would it work fine without DMA? Should we expect a fix any time soon?

andy-shev, 2016-12-08:
@VladimirAkopyan
1. Yes, you may take my branch and compile as usual. No magic or hidden stuff, just `make x86_64_defconfig && make -j8`.
2. a) It will work fine as far as you okay with CPU performance degradation (since each block between 1 ..FIFO_size / 2 character(s) will require to be served through normal interrupt).
2. b) My work is unofficial, no expectations at all, sorry. Of course I would much appreciate any help regarding to hunt the issue.

andy-shev, 2016-12-15:
It seems I found a root cause. But let me test this better and I will submit patches to my branch soon for getting more tests from you.

andy-shev, 2016-12-16:
So, I just have updated my branch. Will very appreciated broad testing!

VladimirAkopyan, 2016-12-16:
Cool! thanks. I've never built a kernel before, but i'll look into it and see how far I can get

htot, 2016-12-16:
I would like to try. But I need to know how toggle gpio mmapped on the Edison Arduino board, then remove mraa from my code. Any hints?

amcduffee, 2016-12-17:
@andy-shev I don't see anymore IRQ #54 traces in my initial checks of the new 64-bit kernel, but have only looked briefly.  I will do a more lengthy test next week.

andy-shev, 2016-12-17:
@htot, from this I barely understand what is the issue you try to resolve?

andy-shev, 2016-12-17:
@amcduffee, will wait for you comprehensive tests, thanks!

htot, 2016-12-20:
@andy-shev https://github.com/htot/hs_uart I need to toggle to 2 gpio pins to a get trigger for the oscilloscope. I am using mmapped to prevent large delays (reduces from 50us to 2 us, relevant when transmitting at 2Mb/s). So need to replace all mraa stuff to be able to test with your kernel. I am particularly interested to test if transmission has the same transmission interchar gap every 45 msec as the standard 3.10.14 kernel.

andy-shev, 2016-12-21:
@htot, got it! So, you just need to use **sysfs** directly or, if it somehow ready, newly provided character device with associated IOCTLS. Here are the links to get a clue: a) [sysfs interface](https://www.kernel.org/doc/Documentation/gpio/sysfs.txt), b) [GPIO user space tools for character device (state of progress unknown)](http://lxr.free-electrons.com/source/tools/gpio/)

htot, 2016-12-21:
@andy-shev Thanks, so you say I could also dig through the mraa source code and use that as an example from here https://github.com/intel-iot-devkit/mraa/blob/master/src/x86/intel_edison_fab_c.c? That has the lines at the top:
// This is an absolute path to a resource file found within sysfs.
// Might not always be correct. First thing to check if mmap stops
// working. Check the device for 0x1199 and Intel Vendor (0x8086)
#define MMAP_PATH "/sys/devices/pci0000:00/0000:00:0c.0/resource0"

andy-shev, 2016-12-21:
@htot, zOMG, I didn't realize that mraa is **so** ugly! Mentioned device is GPIO controller. sysfs priovides standard ABI to communicate with (including all GPIO expanders that found on Edison/Arduino). OTOH pin controller stuff is not available in the user space. I would expect we can do the proper way to use it whenever we will switch to ACPI as resource provider on Edison (U-Boot).

htot, 2016-12-21:
@andy-shev Right now the quick-and-dirty way would be good enough for my testing :-)

amcduffee, 2016-12-27:
@andy-shev This issue appears fixed to me.  Before the holiday weekend I spent about 20 minutes generating activity on the console without any IRQ 54 stack traces appearing.  I also left the Edison module up over the weekend and after ~5 days of up time the dmesg log is still clean.

andy-shev, 2016-12-28:
@amcduffee, cool! So, based on your test I close this issue, thanks!

closed 2016-12-28 by andy-shev

amcduffee, 2016-12-28:
I guess maybe I should have explained how I generated activity.  I was using Minicom to run various commands (dmesg included) that generate a decent amount of text output.  Since I was using Minicom the stdout, of the commands, was going over ttyS2.

Anyhow, I have now also checked the following speeds: 115200, 230400, 921600 and 2000000.  All of them are working as expected.  Also, just for fun I ran the following command at 2000000:
`dd bs=1M count=16 if=/dev/zero | hexdump -Cv`

It completed after ~7 minutes of zeros flying by in Minicom.  No IRQ 54 stack traces on dmesg.

VladimirAkopyan, 2016-12-29:
Amazin!

andy-shev, 2016-12-29:
@amcduffee, yes, thanks! But please check with **/dev/ttyS1**. It's indeed important to me.