andy-shev/linux #6: Wifi not detected

ecc1, 2016-12-03:
My understanding is that wifi should be working on the Edison, but I'm unable to see the device (I'm using an Intel mini-breakout board).  I've tried with the i368_defconfig and others but I must be missing something.

Can someone post a known-working .config file that boots on a mini-breakout and has working wifi?

andy-shev, 2016-12-05:
Please, don't duplicate, though feel free to reopen.
https://github.com/andy-shev/linux/issues/4

I will ask for mini breakout (Currently I'm using only Arduino variant).

closed 2016-12-05 by andy-shev

andy-shev, 2016-12-07:
I commented in #4, but basically the recommendation just to start from the config I'm providing. If it works continue with yours stuff. But I can't reproduce the issue with it. So, it **is** a working one.