andy-shev/linux #7: How to use yocto build for intel eds branch?

sandeep-mvd, 2017-01-16:
Hi,

How can i use eds branch in the yocto environment of intel edison.
The current recipee is using 3.10 kernel. I just replaced the git repository and the branch to use eds branch and it fails to build.
Is there a standard way how I should use this branch on yocto environment? If not, I will try to put my debug information here.

Thank you!

dislick, 2017-01-16:
I would love to know that as well, thank you for your work @andy-shev.

andy-shev, 2017-01-17:
I would try to understand how to do that in the future when I have free time slot. Never used Yocto before.

egorf, 2017-01-23:
Hi!

This recipe is based on the latest official image sources. Unfortunately, Intel's configuration step is a small hack, but otherwise will work OK.

To use it:

* Find the original `linux-externalsrc.bb`. 
* Replace its contents with the ones I've added below
* In the `do_configure` change the `meta-xxxx-xxxx` to whatever is your location of the kernel recipe
* Copy Andy's `i386_defconfig` to the `files` directory next to the recipe. I called it `andy-defconfig`, so either rename it, or change it in the `do_configure`

#### Warning

This will work when building the kernel only(`bitbake linux-externalsrc`). Unfortunately, building the whole image fails during the wi-fi kernel module compilation. Didn't really have much time to poke around with this.

```
LICENSE = "GPLv2"
LIC_FILES_CHKSUM = "file://COPYING;md5=d7810fab7487fb0aad327b76f1be7cd7"

inherit kernel
require recipes-kernel/linux/linux-yocto.inc

FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI = "git://github.com/andy-shev/linux.git;branch=eds;"
SRCREV = "c116b8fd0d9c601a22fe9f94455531f2862c544d"

# Don't use Yocto kernel configuration system, we instead simply override do_configure
# to copy our defconfig in the build directory just before building.
# I agree this is very ad hoc, but maybe it's good enough for our development environment
do_configure() {
  cp "${EDISONREPO_TOP_DIR}/meta-xxxx-xxxx/recipes-kernel/linux/files/andy-defconfig" "${B}/.config"
}

EXTERNALSRC_pn-linux-externalsrc = "${S}"
EXTERNALSRC_BUILD_pn-linux-externalsrc = "${B}"

LINUX_VERSION ?= "3.10"
LINUX_VERSION_EXTENSION = "-edison-${LINUX_KERNEL_TYPE}"

S = "${EDISONREPO_TOP_DIR}/linux-kernel"
B = "${WORKDIR}/${BP}"

# This is required for kernel to do the build out-of-tree.
# If this is not set, most of the kernel make targets won't work properly
# as they'll be executed in the sources
export KBUILD_OUTPUT="${B}"

# The previous line should not be necessary when those 2 are added
# but it doesn't work..
KBUILD_OUTPUT = "${B}"
OE_TERMINAL_EXPORTS += "KBUILD_OUTPUT"

PR = "r2"

COMPATIBLE_MACHINE = "edison"

do_deploy() {
  kernel_do_deploy
  install ${B}/vmlinux ${DEPLOYDIR}/vmlinux
}

do_kernel_configme() {
  echo "skip this option"
}
```

andy-shev, 2017-01-23:
Guys, can we move this issue to **meta-acpi** repository (I recently forked one)?
I dunno how to exactly move the issue, if someone knows, let me know.

andy-shev, 2017-01-23:
Got an answer from tech support that there is no such feature. Please, reopen this bug in **meta-acpi** repository. I'm about to enable issue tracker there.

closed 2017-01-23 by andy-shev

andy-shev, 2018-01-02:
By the way @htot gets full build ready of Yocto based on latest and greates vanilla kernel and U-Boot. Details are here: https://github.com/htot/meta-intel-edison/