andy-shev/linux #8: Problem accessing GPIO pin 23 (SPI) on an Explorer board

PieterGit, 2017-01-23:
Hi @andy-shev 

I am trying to get a TI CCxxxx chip working on an Intel Explorboard (see https://github.com/EnhancedRadioDevices/915MHzEdisonExplorer ) with your eds branch 4.10rc4. 

I now have the `/dev/spidev5.1` file and I also have a `/sys/class/gpio/export` file
```
# ls -al /sys/class/gpio/export
--w------- 1 root root 4096 Jan  1  2000 /sys/class/gpio/export
```

Unfortunately accessing the GPIO pin 23 fails, e.g.:
```
# reset_spi_serial.py
Traceback (most recent call last):
  File "/usr/local/bin/reset_spi_serial.py", line 6, in <module>
    exec(compile(open(__file__).read(), __file__, 'exec'))
  File "/root/src/915MHzEdisonExplorer_SW/spi_serial/scripts/reset_spi_serial.py", line 3, in <module>
    s = spi_serial.SpiSerial()
  File "/root/src/915MHzEdisonExplorer_SW/spi_serial/spi_serial/spi_serial.py", line 7, in __init__
    self.cs0 = m.Gpio(23)
  File "/usr/local/lib/python2.7/dist-packages/mraa.py", line 995, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified
```

This works fine on a Yocto image. Can you explain why I can't access the spi serial pin?

My libmraa version v1.5.1-18-ge386300

This is also discussed on the Intend-to-Bolus gitter, see: https://gitter.im/nightscout/intend-to-bolus?at=588532efd43728124e8ef673 . @ecc1 and @oskarpearson  helped me add some of the necessary kernel parameters. I enabled `CONFIG_I2C_GPIO`, `CONFIG_SPI_GPIO`, `CONFIG_GPIO_SYSFS` on your x86_64_defconfig

Thanks in advance,

Pieter

andy-shev, 2017-01-23:
Regarding to a log on the link I might give a hint: GPIO export works exactly how it's designed in upstream kernel. If *libmraa* doesn't support a **standard** way of doing things it should be fixed.

For me it looks like a completely problem of *libmraa*. I never use that library and never will.

andy-shev, 2017-01-23:
Indeed, my kernel configuration doesn't have GPIO_SYSFS enabled.

andy-shev, 2017-01-23:
Okay, I got what's wrong. Thanks for report.

PieterGit, 2017-01-23:
@andy-shev : That would be great! 
I didn't know linux kernel planning for 4.12-rc1 already exists. 
Thanks for your looking into the problem. If there is anything I can test or debug, please let me know.

VladimirAkopyan, 2017-01-24:
U guys use it for insulin pumps? Holy cow, and I thoughts controlling a greenhouse is risky cause I might damage some salad...

andy-shev, 2017-01-24:
Hold on, I just checked schematics and there is no such GPIO available to the user. Even for the rest I2C 4 pins (I2C1 and I2C6) there is no possibility to switch their mode to something else (in original Edison BSP).

So, back to square 1, i.e. this is clearly 2 bugs: a) official Edison BSP is buggy, b) *libmraa* is crap.

closed 2017-01-24 by andy-shev