andy-shev/linux #10: USB host on Edison

tsaubergine, 2017-04-13:
Hi -

I can get HEAD of https://github.com/andy-shev/linux/tree/eds to boot fine, but USB doesn't seem to work correctly as host (even with an OTG cable and several devices that work using the Intel supplied 3.10 kernel). 

Is this a known limitation? If not, what driver is used for the USB on the Edison. xhci loads automatically for this kernel.

andy-shev, 2017-04-15:
I'm using Edison/Arduino board and don't have such issues. Do you use the same configuration?

tsaubergine, 2017-04-17:
No, I'm using the mini-breakout: https://www.sparkfun.com/products/13025

Could you point me to a working defconfig that you're using?

felipebalbi, 2017-04-18:
The reason is probably because upstream doesn't have support for the OTG block in DWC3. There are people working on it, but it's not ready yet. When you have the physical switch, then it's easier as we can force a port to host/peripheral based on the switch state. Without the physical switch, then we need the OTG interrupt which we're not using yet.

This is, BTW, a guess at this point. I haven't touched that part of the IP yet.

tsaubergine, 2017-04-18:
Thanks, that guess is helpful. I'll get an Arduino board and see what I can learn.

iracigt, 2017-05-03:
I had similar issues with the Edison Arduino board that appeared to stem from a pin responsible for for controlling the onboard 5V regulator not being enabled. We built our own board that did not use this pin and USB works fine. I've done some digging in the SFI tables in an attempt to find the GPIO pin responsible, but have come up empty handed. For reference, the pin in question is PSW, pin 21 on the Hirose connector.

andy-shev, 2017-05-03:
@iracigt 
I'm pretty sure that pin is connected to PMIC (Basin Cove). We have no driver for it in upstream (only power button is supported),

htot, 2017-05-26:
@andy-shev I'm using a Edison Arduino with your kernel and it loads the xhci driver, not the dwc3-pci. Afaik both are built-in the kernel. How do I get it to use the right driver? Kernel command line option?

htot, 2017-05-26:
@andy-shev So, I maybe use the wrong driver and maybe the PSW pin has something to do with this.

But I solved my issue (partly). I am using an evaluation board for the smsc95xx (usb +eth hub). I has a jumper to select bus powered or self powered. I always have it self powered with external power adapter. Now switched to bus powered, and all the leds go on, eth works, usb stick appears. I don't know what that jumper does exactly, because even though 'bus powered' when I pull the adapter, everything turn off. I'll read the docs of the eval board and report back here.

htot, 2017-05-26:
@andy-shev It probably is caused by the PSW pin. I checked the schematics for the EVB9514, it has 2 jumpers: JP1 selects the source of power from either ext or bus. JP22 select the VBUS_DET to sense the bus supplied VCC, or tied to the boards 3.3V. I am now using the combination for permanently connected self powered hub, supplying from ext and disabling VBUS_DET. That means the hub will not see plug-in event, except when powered on. It also means Edison is not powering the USB VCC, so with VBUS_DET enabled the hub does not receive a plug-in event.

PSW on the Edison Arduino of course controls a MIC2039 power switch, which is according to the BOM MIC2039EYMT (active high). The EN is pulled down by a 100k, so when not driven Vbus_out will be off.

mc0rn, 2017-05-27:
I worked around this by porting the USB PHY fix from the original edison linux code. Host mode works great on both the Intel Arduino and Mini Breakout boards running 4.12-rc2.

https://github.com/01org/edison-linux/commit/52b32b192a78de8f1715005bcb6fb4cb928f1e48

andy-shev, 2017-05-27:
@felipebalbi 
Felipe, what do you thing about mentioned change? Can we upstream something like that?
@mc0rn 
Care to send me a pull request, so I will incorporate this into my branch for now?

htot, 2017-05-27:
@andy-shev In the mean time, I'm sure there must be workaround :-) How else did you test the USB else than plugging a USB stick which would need power from the Edison? Can we access the pin from user space?

andy-shev, 2017-05-27:
We have USB hub connected to the board with USB memory stick and network card.

htot, 2017-05-27:
I see, similar situation then. My SMSC eval board set us back €200, almost double the price of the Edison Arduino. The chip itself goes for €2. Rip off.

I'll make the Yocto recipe apply the patch on top of your kernel for now.

tsaubergine, 2017-07-07:
I see that some of these changes have been merged into the latest 4.12 kernel. I tried building this (d88fa47004f0aa7f6290bc8b9665e356a1cd4f5b) but still have no success with USB. Now I don't even have any usb devices (not even the root) in /sys/bus/usb/devices.

If someone has this working with the mini-breakout (@htot?), I would love to have a commit hash and .config that works. Thanks!

htot, 2017-07-07:
No, I don't have the mini breakout. I did backport the patch to Andy's 4.11 kernel (that you can find in my linux/eds-4.11 branch) and it works for me. But tested only on the Edison Arduino board. I branched eds-4.12 but need to check if I can now drop the patch and test if all still works.

htot, 2017-07-08:
@andy-shev Same here, with [d88fa47](https://github.com/htot/linux/commit/d88fa47004f0aa7f6290bc8b9665e356a1cd4f5b) there seems to be no usb. And

`journalctl -b | grep -i dwc
`

returns nothing. But

`root@edison:~# cat /sys/kernel/debug/dwc3.1.auto/mode 
device
`

Not sure if that is correct. Do I need something on the kernel command line to get dwc3 in the right mode?

andy-shev, 2017-07-08:
I think one needs to enable PHY in the config. (TI WL120x?). But best way is to bisect.

tsaubergine, 2017-07-10:
Pardon my ignorance, but I don't know what you mean by "best way is to bisect".

andy-shev, 2017-07-10:
@tsaubergine 
I'm referring to `git bisect` command. It does in O(ln(N)) approach to find the culprit commit if there __was__ a working point.

htot, 2017-07-10:
Acc. @mc0rn it worked at 4.12-rc2. Only 3294 commits since then (minus your 40), needs up to 12 bisects.

I tried adding:

`CONFIG_GENERIC_PHY=y`
`CONFIG_PHY_TUSB1210=y`

to defconfig, but no luck.

Could it not be that dwc3 needs to be put in host or otg mode (now device)? But when I do:
`echo host > /sys/kernel/debug/dwc3.1.auto/mode`
I get:
```------------[ cut here ]------------
 WARNING: CPU: 0 PID: 2449 at /home/ferry/tmp/edison-intel/my/edison-morty/out/linux64/build/tmp/work-shared/edison/kernel-source/kernel/workqueue.c:1442 __queue_work+0x184/0x360

 Modules linked in: i2c_dev iptable_nat nf_nat_ipv4 nf_nat brcmfmac brcmutil spidev spi_pxa2xx_platform spi_pxa2xx_pci pwm_lpss_pci pwm_lpss mmc_block sdhci_pci sdhci led_class mmc_c

 CPU: 0 PID: 2449 Comm: sh Not tainted 4.12.0-edison-standard #2
 Hardware name: Intel Corporation Merrifield/BODEGA BAY, BIOS 542 2015.01.21:18.19.48
 task: f6a92400 task.stack: f524c000
 EIP: __queue_work+0x184/0x360
 EFLAGS: 00010086 CPU: 0
 EAX: 00000000 EBX: f58d0c14 ECX: 00000001 EDX: f58d0c18
 ESI: f6809700 EDI: f6fee100 EBP: f524de74 ESP: f524de58
  DS: 007b ES: 007b FS: 00d8 GS: 0033 SS: 0068
 CR0: 80050033 CR2: 09dba8c8 CR3: 35278000 CR4: 001006d0
 Call Trace:
  queue_work_on+0x1d/0x30
  dwc3_set_mode+0x38/0x40
  dwc3_mode_write+0x80/0xb0
  ? do_filp_open+0x6c/0xb0
  full_proxy_write+0x39/0x70
  ? full_proxy_unlocked_ioctl+0x60/0x60
  __vfs_write+0x22/0x120
  ? security_file_permission+0x31/0xb0
  ? rw_verify_area+0x5c/0x110
  vfs_write+0x93/0x1a0
  SyS_write+0x37/0x80
  do_fast_syscall_32+0x62/0x120
  entry_SYSENTER_32+0x47/0x71
 EIP: 0xb77e9b39
 EFLAGS: 00000246 CPU: 0
 EAX: ffffffda EBX: 00000001 ECX: 09dba8c8 EDX: 00000005
 ESI: 48532d60 EDI: 00000005 EBP: 09dba8c8 ESP: bfe10d08
  DS: 007b ES: 007b FS: 0000 GS: 0033 SS: 007b
 Code: 8b 3e 90 8d 74 26 00 8b 46 04 83 c6 0c 53 8b 55 ec 8b 4d f0 ff d7 8b 3e 5a 85 ff 75 ea 8b 43 04 8d 53 04 39 c2 0f 84 66 ff ff ff <0f> ff 8b 45 f0 8b 00 c6 00 00 8d 65 f4 5b 5e

 ---[ end trace a4c3dc64eff0772c ]---
```

mc0rn, 2017-07-10:
I just checked out and compiled Andy's eds branch and host mode works on the Intel Mini Breakout, Intel Arduino and Sparkfun base board. 

I also checked mainline 4.12 kernel with the same config and it didn't work on any.

Are you looking for host mode only (which this patch fixed) or to re-configure between host,otg on the fly?

tsaubergine, 2017-07-11:
@mc0rn: could you post or link to your config? I didn't have success with the arch/x86/configs/i386_defconfig at Andy's eds branch.

I only need host, not OTG.

Thanks!

mc0rn, 2017-07-11:
@tsaubergine  Here you go
[32bitconfig.txt](https://github.com/andy-shev/linux/files/1137505/32bitconfig.txt)

htot, 2017-07-11:
@mc0rn do you mean that you checked out  [d88fa47](https://github.com/andy-shev/linux/tree/d88fa47004f0aa7f6290bc8b9665e356a1cd4f5b) without any additional patches / changes to the config? Because that is what I do and is not working on the Arduino.

mc0rn, 2017-07-11:
d88fa47 but changed the config for DWC3 USB host mode only. 

CONFIG_USB_DWC3_HOST=y
# CONFIG_USB_DWC3_GADGET is not set
# CONFIG_USB_DWC3_DUAL_ROLE is not set

htot, 2017-07-11:
Thanks, that makes sense, I'll try that tomorrow.

htot, 2017-07-13:
@mc0rn Thanks that worked, thanks.
Acutally I created 2 (what yocto calls) fragments:
```
$ cat usb_phy.cfg 
CONFIG_GENERIC_PHY=y
CONFIG_PHY_TUSB1210=y
```
and 
```
$ cat usb_dwc3.cfg 
CONFIG_USB_DWC3_HOST=y
# CONFIG_USB_DWC3_DUAL_ROLE is not set
```
These are generated by yocto's menuconfig scheme. The first was suggested by Andy, I don't know if that is needed, but it doesn't harm. The 2nd worked. I will commit after some testing, possibly tomorrow.

PS you do know your config above has quite a few differences with Andy's when you do a diff?

floion, 2017-07-14:
@htot how did you force loading of the dwc3-pci driver? Did you deactivate xhci altogether?

htot, 2017-07-14:
I had problems initially with the usb driver, but these were caused by not having updated U-Boot by Andy's version. After the update, this worked fine until breaking as described above and now fixed again.

If you use my layer to build the kernel, you will build a kernel with initramsfs built-in. If you don't want to build/boot the rootfs on the sd card, instead of switch root you can just remain in the initramfs. It has busy box. Some drivers are built into the kernel, some are modules inside the initramfs (as they can not be built-in without disrupting stuff and are needed to get access to the sd card). All the rest modules are loaded from the sd card after that.

As you can see, I haven't changed anything to Andy's kernel (up to now, as he declared a freeze). All my (few) changes are done through patches in the recipe. I built new 4.12 kernel/rootfs yesterday and that works. In my wiki are instructions how to do all that, and if unclear I would be happy to fix that. ftoth at telfort.nl

htot, 2018-04-08:
@felipebalbi Do you know the status of this? I am now on 4.16 and tried with dual role enabled. I had a faulty .config trying to enable g_multi that generated a phy_generic module. And with that for some reason host usb worked. After fixing the config I can't get host mode to work. I got g_multi to work though.

I seems the controller is in device mode all the time. Setting to host (as above) still causes the same crash.

 Of course CONFIG_USB_DWC3_HOST=y still works.

andy-shev, 2018-08-01:
@htot,
I know what the root cause of the traceback you saw when `echo host > mode`.

andy-shev, 2019-02-18:
Latest branch I have consists USB OTG working on Edison/Arduino case. I need to find mini-breakout board (I'm not sure they are still left in our office) and perhaps someone can share (send) Sparkfun base block to test or can test themselves?

andy-shev, 2019-03-11:
Okay, with ACPI enabled U-Boot and kernel with latest patches USB DR works as a charm. I have tested this on Intel Edison mini-breakout board with external power supply and OTG connector / cable.

closed 2019-03-11 by andy-shev