andy-shev/linux #11: Feature request: branch of eds after each linux release

htot, 2017-05-07:
Hello Andy, 

I am working to update meta-intel-edison to poky morty and to incorporate your u-boot and linux repositories. That will hopefully encourage more people to upgrade to your kernel. The good news that it already builds kernel, u-boot and root-fs images! Now I need to prepare to flash and check that it actually boots, so I can upload that work to github.
It would be nice if the resulting images would be based on more or less stable kernels, however your eds HEAD tracks kernel rc's with your patches applied. Would it be possible to create a branch with the latest released kernel (with your patched applied)?
Currently I am cloning a07f946, but that will break of course when you rebase to the next kernel rc.

Ferry

andy-shev, 2017-05-08:
You may start just from v4.11 vanilla. My branch has only bluetooth on top of it. So, I won't do any tags — use vanilla directly!

closed 2017-05-08 by andy-shev

htot, 2017-05-08:
Alright, fair enough. I'll create branches myself then while tracking your repo. And put features that you are not providing by default (usb ethernet driver, ftdi serial driver) in patches in the meta-layer.

andy-shev, 2017-05-08:
Yes, it would be best choice for you! v4.11 vanilla is quite functional on Edison, and you may take as a base. Two patches (that as I said will be part of v4.12-rc1) willenable Bluetooth. Everything else it's your own configuration and stuff.

On the contrary U-Boot repo has tags and I'll continue that until minimum support will be there in upstream.