andy-shev/linux #12: 4.11 does not load rootfs from mmc

htot, 2017-05-18:
I have the u-boot edison-v2017.05 and built the 32bits 4.11 version of the kernel, loaded that to the FAT formated part 9 and used: 
```
fw_setenv bootcmd_edsboot 'setenv bootargs ${bootargs_edsboot}; run load_edsboot; run boot_edsboot'
fw_setenv bootargs_edsboot 'console=tty1 console=ttyS2,115200n8 rootwait root=PARTUUID=012b3303-34ac-284d-99b4-34e03a2335f4 rootfstype=ext4'
fw_setenv load_edsboot 'load mmc 0:9 0x100000 bzImage'
fw_setenv boot_edsboot 'zboot 0x100000'
```

The kernel boots fine up to the point:
[    2.432544] md: Waiting for all devices to be available before autodetect
[    2.439398] md: If you don't use raid, use raid=noautodetect
[    2.446107] md: Autodetecting RAID arrays.
[    2.450279] md: autorun ...
[    2.453122] md: ... autorun DONE.
[    2.456551] Waiting for root device PARTUUID=012b3303-34ac-284d-99b4-34e03a2335f4...

And sits there (no watchdog reset). The partition is the original rootfs that is booted by the yocto kernel, on the mmc.

andy-shev, 2017-05-22:
Hmm... I remember I tried once to do such, though I don't remember results. When I have time to check I will inform you here. Not much time for this anymore, sorry.

htot, 2017-05-22:
Yes, I believe you explained me before I need an initramfs, load some drivers, mount the real rootfs and then switch_root.
Today I managed to build such an initramfs. I only included mmc_block and mmc_core, using your exact kernel (configs and all, as it was when 4.11 was released).  I modprobed these manually but didn't see anything appear with dmesg. I don't have systemd (dont know if I need that), there is udev, but that doesn't seem to do much right now.
Would you know which modules I need to insert and which device should appear to get acces to the mmc (or external sdcard would be fine for now too)? I will post back here when I have a solution and try to close this issue myself if I can.

htot, 2017-05-22:
It's my lucky day.  I have yocto building your kernel and create .deb packages. It then installs these to the initramfs. I was missing - of course - the sdhci modules. The module packages installed into the rootfs now are: `"kernel-module-mmc-block kernel-module-mmc-core kernel-module-sdhci kernel-module-sdhci-acpi kernel-module-sdhci-pci"`

That changes everything.
`sdhci: Secure Digital Host Controller Interface driver
sdhci: Copyright(c) Pierre Ossman
sdhci-pci 0000:00:01.0: SDHCI controller found [8086:1190] (rev 1)
mmc0: SDHCI controller on PCI [0000:00:01.0] using ADMA
sdhci-pci 0000:00:01.2: SDHCI controller found [8086:1190] (rev 1)
cat (1257) used greatest stack depth: 6872 bytes left
mmc1: SDHCI controller on PCI [0000:00:01.2] using ADMA

Waiting for removable media...
  30

sdhci-pci 0000:00:01.3: SDHCI controller found [8086:1190] (rev 1)
mmc2: SDHCI controller on PCI [0000:00:01.3] using ADMA
mmc0: new DDR MMC card at address 0001
logger (1305) used greatest stack depth: 6796 bytes left
mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
mmcblk0: mmc0:0001 H4G1d 3.64 GiB 
mmc2: queuing unknown CIS tuple 0x80 (6 bytes)
mmcblk0boot0: mmc0:0001 H4G1d partition 1 4.00 MiB
mmcblk0boot1: mmc0:0001 H4G1d partition 2 4.00 MiB
mmc2: queuing unknown CIS tuple 0x80 (2 bytes)
mmcblk0rpmb: mmc0:0001 H4G1d partition 3 4.00 MiB
 mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10
mmc2: queuing unknown CIS tuple 0x80 (4 bytes)
random: fast init done
mmc2: new ultra high speed DDR50 SDIO card at address 0001
EXT4-fs (mmcblk0p5): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (mmcblk0p5): couldn't mount as ext2 due to feature incompatibilities
EXT4-fs (mmcblk0p8): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (mmcblk0p10): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (mmcblk0p10): couldn't mount as ext2 due to feature incompatibilities
EXT4-fs (mmcblk0p8): couldn't mount as ext2 due to feature incompatibilities
EXT4-fs (mmcblk0p8): recovery complete
EXT4-fs (mmcblk0p5): warning: mounting unchecked fs, running e2fsck is recommended
EXT4-fs (mmcblk0p8): mounted filesystem with ordered data mode. Opts: (null)
mount (1497) used greatest stack depth: 6664 bytes left
EXT4-fs (mmcblk0p5): mounted filesystem without journal. Opts: (null)

  29 

EXT4-fs (mmcblk0p10): mounted filesystem with ordered data mode. Opts: (null)
udevd (989) used greatest stack depth: 6068 bytes left
  
28  27  26  25  24  23  22  21  20  19  18  17  16  15  14  13  12  11  10  9  8  7  6  5  4  3  2  1  0...

Mounted filesystems
/dev/mmcblk0p8 on /run/media/mmcblk0p8 type ext4 (rw,relatime,data=ordered)
/dev/mmcblk0p5 on /run/media/mmcblk0p5 type ext4 (rw,relatime,block_validity,delalloc,barrier,user_xattr,acl)
/dev/mmcblk0p7 on /run/media/mmcblk0p7 type vfat (rw,relatime,gid=6,fmask=0007,dmask=0007,allow_utime=0020,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p9 on /run/media/mmcblk0p9 type vfat (rw,relatime,gid=6,fmask=0007,dmask=0007,allow_utime=0020,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p10 on /run/media/mmcblk0p10 type ext4 (rw,relatime,data=ordered)
Available block devices
major minor  #blocks  name

 179        0    3817472 mmcblk0
 179        1       2048 mmcblk0p1
 179        2       1024 mmcblk0p2
 179        3       2048 mmcblk0p3
 179        4       1024 mmcblk0p4
 179        5       1024 mmcblk0p5
 179        6      24576 mmcblk0p6
 179        7      32768 mmcblk0p7
 259        0    1572864 mmcblk0p8
 259        1     786432 mmcblk0p9
 259        2    1392623 mmcblk0p10
 179       24       4096 mmcblk0rpmb
 179       16       4096 mmcblk0boot1
 179        8       4096 mmcblk0boot0
Cannot find rootfs.img file in /run/media/mmcblk0p10 /run/media/mmcblk0p5 /run/media/mmcblk0p7 /run/media/mmcblk0p8 /run/media/mmcblk0p9 , dropping to a shell

sh: can't access tty; job control turned off
`
So all I need to do now is fix the init script. After I clean up the mess a bit I will push the recipies to my repository. Closing this issue.

closed 2017-05-22 by htot

htot, 2017-05-31:
I fixed my initramfs init script and my rootfs boots nicely (well with a small hickup as it expects nfsd, but the kernel module is not enabled in defconfig). Wow