andy-shev/linux #14: ttyS1 on edison_arduino won't work (linux 4.11)

htot, 2017-06-25:
I have tried this with `screen /dev/ttyS1 115200` (running on edison) and with a small program that writes 65 bytes out and reads back whatever is in the buffer (currently read exactly 0 bytes).  I have RX and TX looped back of course.

External driver direction set with:
```
# 0 - RX  
# rx (input) is on pin 130, level translator direction must be set on 248
echo 248 > /sys/class/gpio/export 
echo low > /sys/class/gpio/gpio248/direction  # set to input
echo 216 > /sys/class/gpio/export 
echo in > /sys/class/gpio/gpio216/direction  # no pullup

# 1 - TX  
# tx (output) is on 131, level translator direction must be set on 249
echo 249 > /sys/class/gpio/export
echo high > /sys/class/gpio/gpio249/direction  # set to output
echo 217 > /sys/class/gpio/export  
echo in > /sys/class/gpio/gpio217/direction # no pullup
```
I can not set direction on 130/131 as that claims them for gpio, so I assume hsu driver take care of pinctrl and pinmux  (once claimed I don't know how to release back to hsu, except by reboot).

Of course ttyS2 is used for the console and works fine. According to:
```
cat /sys/kernel/debug/pinctrl/pinctrl-merrifield/pins
pin 121 (GP130_UART_1_RX) mode 1 0x00003001
pin 122 (GP131_UART_1_TX) mode 1 0x00003001
pin 125 (GP134_UART_2_RX) mode 1 0x00003121
pin 126 (GP135_UART_2_TX) mode 1 0x00003121
```
all should be in mode 1, however there is a difference in the other bits.

What am I doing wrong here?

andy-shev, 2017-06-25:
I'm travelling right now, so, let me check this lately.
Pay attention that EOL is set for the Edison, so, these kind of tasks now are at lowest priority.

htot, 2017-06-25:
Yes, the EOL for the boards (breakout/arduino) doesn't surprise me, but for the compute module and no alternative announced really pisses me off.

htot, 2017-06-27:
The problem in the end was caused by my script where I had disabled the final TRI_STATE_ALL = high (due to enumeration problems I had earlier).

Note to self: 

TRI_STATE_ALL controls the INH pin on the 74LVC2G53 analog multiplexers, but also cuts power to the 74LVC1T45 level translators (by the V_SHIELD_SW signal, a global, and if I may say so, the use of globals in schematics is just as despicable as in software). So effectively by default all analog/digital pin are tri-stated (as the signal name already says).

So it is really important to enable TRI_STATE_ALL like so (f.i.):
```
# TRI_STATE_ALL signal is controlled by GPIO 214                                                                                                     
echo 214 > /sys/class/gpio/export    
echo high > /sys/class/gpio/gpio214/direction                                                                                                                
```

closed 2017-06-27 by htot