andy-shev/linux #16: Kernel won't boot

floion, 2017-07-14:
Hi guys,
I am using git://github.com/htot/linux.git , branch eds-4.11 which should be pretty much the same kernel as the one from this repo.
However, the kernel won't boot. It just hangs at "Starting kernel ..." forever.
Has anybody seen this with this kernel?
Thanks

floion, 2017-07-14:
Just tried with d88fa47004f0aa7f6290bc8b9665e356a1cd4f5b and I get the same thing

floion, 2017-07-14:
@andy-shev have you ever seen this?

AurelienBallier, 2017-07-14:
Use: console=tty1 console=ttyS2,115200n8 loglevel=7

floion, 2017-07-14:
Thanks @AurelienBallier , I have eyes now :)
So the tty has changed for this kernel? I was expecting ttyMFD2 to be exposed as the console still. Doesn't matter too much if it is not exposed anymore.
I can now see the subsequent error.

AurelienBallier, 2017-07-14:
Yes serial interfaces have "standard" names now ttyS0, ttyS1 and ttyS2.

floion, 2017-07-14:
I see, thanks for the info

andy-shev, 2017-07-14:
Thank you @AurelienBallier for taking care about the issue. Closing as invalid based on help given.

closed 2017-07-14 by andy-shev