andy-shev/linux #17: SST probe error

svenschwermer, 2017-10-15:
Hi! When setting `CONFIG_SND_MFLD_MACHINE=y` the kernel tries to probe the SST PCI device `intel_sst_probe`, but fails in `sst_context_init` because platform data (`pdata`) is not set in the driver context. I guess, this needs to be done by a platform driver (sst-mfld-platform-pcm maybe?) or by a SFI driver (matching `mrfld_sst`) perhaps. I see that the platform data is being set in sst_acpi.c but as of now I am running a non-ACPI u-boot. Is this resolved with u-boot from the acpi branch? Or where's the appropriate spot to set the platform data? I see that you've done a few changes to the SST sources, Andy, did you get this to work?

andy-shev, 2017-10-15:
Unfortunately I have been having no time to continue SST enabling. What I heard is there is an activity to bring SOF (Sound Open Firmware) support to the drivers soon. It will change the driver structure as well.
So, I would recommend to look into that direction.

The best person to ask are Liam and Pierre from Intel who are doing ASoC stuff. (You may find them in alsa-devel@ mailing list)

P.S. And no, ACPI branch has no bits for SST. It would be really helpful if someone can eventually help with SST enabling. (I have even RT5640 codec daughter board on my table for half a year, but... no time to accomplish)

svenschwermer, 2017-10-15:
Alright, I'll look into that direction. I'd seen this entry in the MCRS provided by u-boot so I thought I would ask:
```
/* SST Memory Region */
DWordMemory(ResourceProducer, PosDecode, MinFixed, MaxFixed,
    Cacheable, ReadWrite,
    0x00000000, 0x05e00000, 0x05ffffff, 0x00000000,
    0x00200000, , , SST0)
```

andy-shev, 2017-10-15:
Ah, without that one you didn't get device enumerated at all, but it doesn't describe device per se.

andy-shev, 2017-10-15:
By the way I would recommend to find most closest hardware (laptop) and acquire DSDT table from it. Then some bits can be used to create an excerpt for Merrifield (ASuS T100 or alike perhaps).

andy-shev, 2018-07-18:
@pbossart

andy-shev, 2025-06-08:
As of today the platform should run SOF based solution. We probably leave the SST driver abandon.

closed 2025-06-08 by andy-shev