andy-shev/linux #18: eds branch changed?

floion, 2017-11-17:
Hi,
we are using this kernel and been using this revision: 54f9552249b720ef9d33a7eb9e0d1f8f53ce1025
However, today I see this revision is not in the git tree anymore. Was there a git rewrite on your side?

Thank you

floion, 2017-11-17:
Apparently there was a git rewrite due to some change (merge ? ) and now kernel is 4.14.
We were relying on 4.13

floion, 2017-11-17:
Will every merge with the vanilla kernel make this sort of git re-write?

htot, 2017-11-17:
Hello @floion Andy rebases (I think currently only if he has time) each release candidate. That breaks your reference to commit number. To prevent that from happening I pull from him on each kernel release and create a branch. You will find what you need here https://github.com/htot/linux/tree/eds-4.13 (currently 4.11 - 4.14)

htot, 2017-11-17:
I am not adding my own commits on top of these, instead put my own modifications as a fragment in the recipe.

floion, 2017-11-17:
Thanks @htot

andy-shev, 2017-11-18:
Since the minimum needed for Edison to work is in vanilla (starting from v4.12) I'm rebasing only to official releases. @htot is right, the best approach is to use whatever base you want (v4.13 for example) and take patches on top of that if needed. Just a side note, that my tree is an experimental field for the stuff. No one should rely on the content of it. Thanks for understanding.

closed 2017-11-18 by andy-shev

floion, 2017-11-18:
Thanks for the input

htot, 2017-11-20:
So, if you are using Yocto you could either take an official vanilla kernel and add the 40 or so patches from Andy + your own as kernel fragments. Or you take the branch you need from my repo which are just a snapshot of Andy's after each kernel release and add your own kernel fragments to that. My snapshots don't get rewritten and I just keep them there so my recipe doesn't break and I can easily can go back to a previous kernel if needed.