andy-shev/linux #19: unloading SDIO kernel module

farfromrefug, 2017-12-12:
Recently i stumble upon the issue where some Edison GPIOs where could not write on them because they were reserved for the SD:
https://communities.intel.com/thread/77677

I know i can compile a kernel without that module. But i would prefer to be able to unload /load it at runtime.
I don't enought about kernels to actually know where to look to see if this kernel allows this.
Can you guide me?

Thanks

andy-shev, 2017-12-12:
So, the kernel allows user to *unload* module (means that *all* devices will be gone which are supported by the driver in question) if there is no active user (of *all* of them), or *unbind* certain device from the driver — condition is the same, no active user of it.
There is a good article to start with if you are interested in the latter mechanism https://lwn.net/Articles/143397/

andy-shev, 2019-02-18:
@farfromrefug, is everything clarified?

farfromrefug, 2019-02-19:
@andy-shev thanks yes it s  clear now

andy-shev, 2019-02-19:
Closed based on comment from @farfromrefug.

closed 2019-02-19 by andy-shev