andy-shev/linux #20: Bluetooth broken on 4.14

htot, 2017-12-27:
Looks like bt support broke in 4.14. I now get:
```
root@edison:~# /usr/bin/btattach  -B /dev/ttyS0 -P bcm -S 3000000
Attaching Primary controller to /dev/ttyS0
Switched line discipline from 0 to 15
Failed to set protocol: Protocol not supported
No controller attached
```
If I remember correctly for sure in 4.12, maybe also 4.13.

andy-shev, 2017-12-28:
Can you revert *commit 276c87054751 ("x86/platform/intel-mid: Make 'bt_sfi_data' const")* and check if it works again?

andy-shev, 2017-12-28:
For the reference: https://github.com/htot/meta-intel-edison/issues/13

andy-shev, 2017-12-28:
One more thing: there is new dependency to Serial Device Bus, which needs to be selected by `CONFIG_SERIAL_DEV_BUS=y` in kernel configuration. I'm going to update my **eds** branch with above fixes.

htot, 2017-12-28:
Could it be we need a fragment:
```
CONFIG_BT_BCM=m
CONFIG_BT_HCIUART_SERDEV=y
CONFIG_BT_HCIUART_H4=y
# CONFIG_BT_HCIUART_NOKIA is not set
# CONFIG_BT_HCIUART_LL is not set
CONFIG_BT_HCIUART_BCM=y
CONFIG_SERIAL_DEV_BUS=y
# CONFIG_SERIAL_DEV_CTRL_TTYPORT is not set
```

htot, 2017-12-28:
I discovered your patch e6a575bcc32b621f107f1ac0c87e69154c787ab7 to late. The fragment above is suggested by menuconfig. Not sure why it is different, but it seems to enable the configs you need and a bit more (more than needed?). Starting bt gives:
```
root@edison:~# systemctl start bluetooth_attach
root@edison:~# [  144.181063] Bluetooth: HCI UART driver ver 2.3
[  144.185560] Bluetooth: HCI UART protocol H4 registered
[  144.192225] hci_bcm hci_bcm: BCM irq: 441
[  144.196303] hci_bcm hci_bcm: hci_bcm device registered.
[  144.205472] Bluetooth: HCI UART protocol Broadcom registered
[  144.363365] Bluetooth: hci0: BCM: chip id 82
[  144.370296] Bluetooth: hci0: BCM: features 0x2f
[  144.398305] Bluetooth: hci0: BCM43341B0
[  144.402222] Bluetooth: hci0: BCM43341B0 (002.001.014) build 0000
[  145.698117] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  145.703488] Bluetooth: BNEP filters: protocol multicast
[  145.708805] Bluetooth: BNEP socket layer initialized
[  148.792848] Bluetooth: hci0: BCM (002.001.014) build 0176
[  148.969243] Bluetooth: RFCOMM TTY layer initialized
[  148.974236] Bluetooth: RFCOMM socket layer initialized
[  148.979503] Bluetooth: RFCOMM ver 1.11
```

htot, 2017-12-28:
Your suggested patches indeed fix the problem. I added as patches to the yocto-linux recipe with c788c20abbb3c8f9e30265e206780befcf9915a5 and will revert that when 4.15 is released.
Thanks!

closed 2017-12-28 by htot