andy-shev/linux #21: Reboot problem on Edison with 3.3V supply

spanceac, 2018-02-09:
Hi,

We have a custom board that supplies the Edison module with 3.3V(VSYS voltage).
We are using the 4.13.0 kernel, compiled from this git repo.
When we give the reboot command, the system hangs.

The only log is the one bellow, without PSH kernel and u-boot starting anymore:

root@d29029e:~# reboot
[ 6213.027804] reboot: Restarting system

If we put the Edison module in the mini breakout board, reboot works fine. The mini breakout board supplies the Edison module with 4.4V.

Also, if we flash our Edison module with 3.10.98 kernel for edison, reboot works even on the custom hardware.

Since the old kernel had source code for Merrifield PMU, I'm thinking that lacking the PMU code in the 4.13.0 kernel could causes the reboot problem.

Do you think this could be the case?

Thanks!

taenadil, 2018-02-15:
Hello,

We have also a custom board and we have the same problem.
We use the kernel 4.15 and u-boot from Andy repo's.
For the OS we used CentOS 7 (64bits).
When we reboot the system it restarts to the micro kernel of the Edison and does not load u-boot.

But if you restart with the OTG cable connected to the Edison, no problem...

spanceac, 2018-02-15:
Hi,

I found a fix for this problem, by replacing the IPCMSG_COLD_REBOOT with IPCMSG_COLD_RESET, as the command sent to SCU at reboot time.

You can see the patch here: https://github.com/resin-os/meta-edison-bsp/blob/master/recipes-kernel/linux/files/0003-Replace-reset-type-command-for-SCU.patch

taenadil, 2018-02-15:
Oh nice
After several tries; it works!!
A big thank you to you

andy-shev, 2018-02-16:
@spanceac 
Hmm... Looking into original Edison BSP we may find

       if (intel_scu_ipc_fw_update()) {
               pr_debug("intel_scu_fw_update: IFWI upgrade failed...\n");
       }
       if (force_cold_boot)
               rpmsg_send_generic_simple_command(IPCMSG_COLD_BOOT, 0);
       else
               rpmsg_send_generic_simple_command(IPCMSG_COLD_RESET, 0);

OTOH, the difference of the command sent to SCU were before for Lincroft vs. Penwell. Seems Penwell requires COLD_BOOT. 

In any case, I have Penwell here, so, I can test your patch on both Tangier and Penwell SoCs.
Just add proper commit message and send directly upstream (Cc'ing x86 maintainers and me).

Thanks!

spanceac, 2018-02-19:
I'll prepare a patch for upstream.

Regards!

closed 2018-12-29 by spanceac