andy-shev/linux #26: SoC_DT0 and SoC_DT1 (intel temperature sensors) not available

razvan-becheriu, 2019-07-02:
Hi,

The intel temperature drivers are not available in thud version.
I have tried to enable the drivers in the kernel with no success.
Can you point out what are the kernel configuration parameters I need to enable/disable so that I can access the intel temperature sensors trough /sys/class/thermal/ interface?

Thank you,
Razvan

andy-shev, 2019-07-03:
In the original BSP the *drivers/thermal/intel_mrfl_thermal.c* provides such functionality. As far as I know there is no similar functionality provided in vanilla kernel. Note, the vanilla should support rather ACPI approach to use DTS.

razvan-becheriu, 2019-07-03:
I have found that using hwmon you can actually get the temperature:
cat /sys/class/hwmon/hwmon0/temp2_label
Core 0
cat /sys/class/hwmon/hwmon0/temp2_input
55000
cat /sys/class/hwmon/hwmon0/temp3_label
Core 1
cat /sys/class/hwmon/hwmon0/temp3_input
56000

this works for me.
you can close this issue if it is of no interest for you either.

htot, 2019-07-03:
What modules do you need to get this?

andy-shev, 2019-07-04:
> What modules do you need to get this?

It looks like IIO hwmon support.

closed 2019-07-04 by andy-shev