andy-shev/linux #31: eem gadget works only after cold reboot

htot, 2021-03-23:
This was with your configurations rebased on 5.10.24 (LTS) https://github.com/edison-fw/linux/commits/eds-acpi-5.10.24
I have bitbake apply 2 other (relevant) patches:
```
SRC_URI_append = " file://0001-HACK-do-not-enumerated-SFI-devices.patch"
SRC_URI_append = " file://0001-usb-dwc3-pci-Enable-dis_uX_susphy_quirk-for-Intel-Me.patch"
```
I'll highlight the interesting bits and have some conclusions at the end:

### This is to test eem ethernet gadget (USB 2).
**In these tests I leave edison-arduino switch in gadget mode. Communication with edison is through console (USB 1)**

1) Cold boot (press Edison reset button 7 sec), with USB 2 unplugged, wait for prompt, login, plug USB 2

```
root@yuna:~# connmanctl services
*AO Wired                gadget_aabbccddeef1_usb
...

root@yuna:~# ifconfig
...

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500  metric 1
        inet 10.42.0.221  netmask 255.255.255.0  broadcast 10.42.0.255
        inet6 fe80::a8bb:ccff:fedd:eef1  prefixlen 64  scopeid 0x20<link>
        ether aa:bb:cc:dd:ee:f1  txqueuelen 1000  (Ethernet)
        RX packets 45  bytes 7074 (6.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43  bytes 6173 (6.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

root@yuna:~# iperf3 -s
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 10.42.0.1, port 50840
[  5] local 10.42.0.221 port 5201 connected to 10.42.0.1 port 50842
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  23.1 MBytes   193 Mbits/sec
[  5]   1.00-2.00   sec  24.1 MBytes   202 Mbits/sec
[  5]   2.00-3.00   sec  24.4 MBytes   205 Mbits/sec
[  5]   3.00-4.00   sec  24.4 MBytes   205 Mbits/sec
```

On host (run 2 min, at 200Mb/sec Edison doesn't crash in other cases often does)
```
ferry@delfion:~$ iperf3 -c 10.42.0.221 -t 0
Connecting to host 10.42.0.221, port 5201
[  5] local 10.42.0.1 port 50842 connected to 10.42.0.221 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  26.7 MBytes   224 Mbits/sec    0   1.12 MBytes
[  5]   1.00-2.00   sec  23.8 MBytes   199 Mbits/sec   49   1.13 MBytes
[  5]   2.00-3.00   sec  23.8 MBytes   199 Mbits/sec    0   1.25 MBytes
[  5]   3.00-4.00   sec  25.0 MBytes   210 Mbits/sec    0   1.34 MBytes
[  5]   4.00-5.00   sec  23.8 MBytes   199 Mbits/sec    0   1.40 MBytes
[  5]   5.00-6.00   sec  25.0 MBytes   210 Mbits/sec   14   1.04 MBytes
...
[  5] 157.00-158.00 sec  0.00 Bytes  0.00 bits/sec    0   1.21 MBytes
[  5] 158.00-159.00 sec  0.00 Bytes  0.00 bits/sec    0   1.21 MBytes
^C[  5] 159.00-159.32 sec  0.00 Bytes  0.00 bits/sec    0   1.21 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-159.32 sec  3.41 GBytes   184 Mbits/sec  294             sender
[  5]   0.00-159.32 sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated
```

2) kill iperf3 on Edison, unplug USB 2, wait 5 sec, plug USB 3

```
root@yuna:~# connmanctl services
[... 111 lines cut ...]
```
root@yuna:~# connmanctl enable gadget
ferry@delfion:~$ iperf3 -c 10.42.0.221 -t 0
Connecting to host 10.42.0.221, port 5201
[  5] local 10.42.0.1 port 40422 connected to 10.42.0.221 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  25.6 MBytes   215 Mbits/sec    0   1.10 MBytes       
[  5]   1.00-2.00   sec  25.0 MBytes   210 Mbits/sec   89   1.16 MBytes       
[  5]   2.00-3.00   sec  23.8 MBytes   199 Mbits/sec    0   1.28 MBytes       
[  5]   3.00-4.00   sec  23.8 MBytes   199 Mbits/sec    0   1.38 MBytes       
[  5]   4.00-5.00   sec  25.0 MBytes   210 Mbits/sec   11   1.02 MBytes       
[  5]   5.00-6.00   sec  25.0 MBytes   210 Mbits/sec    0   1.09 MBytes       
[  5]   6.00-7.00   sec  23.8 MBytes   199 Mbits/sec    0   1.14 MBytes       
^C[  5]   7.00-7.39   sec  10.0 MBytes   214 Mbits/sec    0   1.16 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-7.39   sec   182 MBytes   206 Mbits/sec  100             sender
[  5]   0.00-7.39   sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated
```
### Conclusion

- On cold boot dwc3 eem gadget works fine. 
- Unplug/replug messes up the driver or the hardware, warm reboot does not reset this behavior.
- In working state, disabling the eem gadget with connmanctl (I believe this to be the same as ` ip link set dev usb0 down`), plug/replug followed by enabling again, eem gadget works.

htot, 2021-03-30:
I'm dropping trace and regdump after a cold and a warm boot captured according to instructions here: https://www.kernel.org/doc/html/latest/driver-api/usb/dwc3.html#reporting-bugs

[boot.zip](https://github.com/andy-shev/linux/files/6232410/boot.zip)

htot, 2021-04-02:
While the above traces in boot.zip allow compare which regs not correctly initialized on warm boot, I have also captured traces of unplug/plug. Here kernel is 5.10.27 (LTS), edison cold booted with USB cable plugged. Loggin via console, verified connection **inet 10.42.0.221  netmask 255.255.255.0**
```
root@yuna:~# ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536  metric 1
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 44  bytes 3412 (3.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 44  bytes 3412 (3.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500  metric 1
        inet 10.42.0.221  netmask 255.255.255.0  broadcast 10.42.0.255
        inet6 fe80::a8bb:ccff:fedd:eef1  prefixlen 64  scopeid 0x20<link>
        ether aa:bb:cc:dd:ee:f1  txqueuelen 1000  (Ethernet)
        RX packets 61  bytes 8416 (8.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 54  bytes 7779 (7.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=-28669<UP,BROADCAST,MULTICAST,DYNAMIC>  mtu 1500  metric 1
        ether 78:4b:87:ad:4d:70  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Then trace unplug. Then trace plug. 

[plug-unplug.zip](https://github.com/andy-shev/linux/files/6250924/plug-unplug.zip)

htot, 2021-04-05:
Now, with host updated to ubuntu kernel ppa 5.12.0-051200rc5-generic and edison to 5.12.0-rc5-edison-acpi-standard vanilla + 2 patches appearing in rc6:
 * "usb: dwc3: gadget: Clear DEP flags after stop transfers in ep disable"
 * "usb: dwc3: pci: Enable dis_uX_susphy_quirk for Intel Merrifield"
plus one from https://github.com/andy-shev/linux/commits/eds-acpi
 * "TODO: driver core: Break infinite loop when deferred probe can't be satisfied"
captured on good and one bad connection, plus logs on the host side see journalctl-plus-comments.txt.
[5.12-rc5.zip](https://github.com/andy-shev/linux/files/6260614/5.12-rc5.zip)

htot, 2021-04-12:
I added another 3:
- usb: dwc3: Fix DRD mode change sequence following programming guide
- usb: dwc3: Trigger a GCTL soft reset when switching modes in DRD
- usb: dwc3: gadget: increase BESL baseline to 6

The latter seems to solve the throttling, while the first 2 help to make switching from host -> device mode work (better).

htot, 2021-04-17:
I retested usb-next (based 5.12-rc7) + "[PATCH v3] usb: dwc3: core: Do core softreset when switch mode". No throttling, but 
- if I flip the switch (or unplug/plug the cable) only once, on the PC side I get:
```
kernel: usb 1-5: new high-speed USB device number 18 using xhci_hcd
kernel: usb 1-5: New USB device found, idVendor=1d6b,
idProduct=0104, bcdDevice= 1.00 kernel: usb 1-5: New USB device
strings: Mfr=1, Product=2, SerialNumber=3 kernel: usb 1-5: Product:
USBArmory Gadget kernel: usb 1-5: Manufacturer: USBArmory kernel:
usb 1-5: SerialNumber: 0123456789abcdef kernel: usb 1-5: can't set
config #1, error -110
```
In the zip file 1 more trace of just the switch from host->gadget
[5.12-rc7+usb-next.zip](https://github.com/andy-shev/linux/files/6329600/5.12-rc7%2Busb-next.zip)

htot, 2021-04-17:
regdump in zip files empty, resend..
[regdump.txt](https://github.com/andy-shev/linux/files/6329639/regdump.txt)

htot, 2021-04-20:
After flipping the first time the trace is empty see [first-switch.zip](https://github.com/andy-shev/linux/files/6346271/first-switch.zip)

I found the gadget configuration script was not called, which normally gets called due to a udev rule:

ACTION=="add", KERNEL=="dwc3.0.auto", SUBSYSTEMS=="udc", ATTRS{state}=="not attached", RUN+="/usr/bin/conf-gadget.sh"

So I retried and see with `udevadm monitor`

flipping the switch from host->gadget
```
KERNEL[51.824914] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/enp0s17u1u1/queues/rx-0 (queues)
KERNEL[51.825682] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/enp0s17u1u1/queues/tx-0 (queues)
KERNEL[51.826226] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/enp0s17u1u1 (net)
KERNEL[51.836041] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:003/usb-001:003:01 (mdio_bus)
KERNEL[51.836709] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:003/usb-001:003:01 (mdio_bus)
KERNEL[51.837342] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:003 (mdio_bus)
KERNEL[51.837763] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0 (usb)
KERNEL[51.838116] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0 (usb)
KERNEL[51.873712] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1 (usb)
KERNEL[51.874000] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1 (usb)
KERNEL[51.874207] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1:1.0 (usb)
KERNEL[51.874431] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1:1.0 (usb)
KERNEL[51.897175] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1 (usb)
KERNEL[51.897486] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1 (usb)
```
stopped capture tracepoints here, then switch back to host
```
KERNEL[253.214406] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1 (usb)
KERNEL[253.263305] change   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1 (usb)
KERNEL[253.263687] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1:1.0 (usb)
KERNEL[253.328354] bind     /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1:1.0 (usb)
KERNEL[253.328734] bind     /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1 (usb)
KERNEL[253.699341] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1 (usb)
KERNEL[253.744911] change   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1 (usb)
KERNEL[253.745804] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0 (usb)
KERNEL[253.805307] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:005 (mdio_bus)
KERNEL[253.812978] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:005/usb-001:005:01 (mdio_bus)
KERNEL[253.814318] bind     /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:005/usb-001:005:01 (mdio_bus)
KERNEL[253.815386] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0 (net)
KERNEL[253.815552] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0/queues/rx-0 (queues)
KERNEL[253.815778] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0/queues/tx-0 (queues)
KERNEL[253.825279] bind     /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0 (usb)
KERNEL[253.825667] bind     /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1 (usb)
```
switch to gadget again
```
KERNEL[314.212144] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0/queues/rx-0 (queues)
KERNEL[314.212473] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0/queues/tx-0 (queues)
KERNEL[314.214691] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0/net/eth0 (net)
```
extcon event didn't show the first time
```
KERNEL[314.238385] change   /devices/pci0000:00/0000:00:13.0/INTC100E:00/mrfld_bcove_pwrsrc/extcon/extcon0 (extcon)
KERNEL[314.238677] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:005/usb-001:005:01 (mdio_bus)
KERNEL[314.238863] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:005/usb-001:005:01 (mdio_bus)
KERNEL[314.239015] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/mdio_bus/usb-001:005 (mdio_bus)
KERNEL[314.239205] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usb1/1-1/1-1.1/1-1.1:1.0 (usb)
[... cut ...]
KERNEL[314.647375] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto/usbmon/usbmon1 (usbmon)
KERNEL[314.647816] unbind   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto (platform)
KERNEL[314.648143] remove   /kernel/software_nodes/node1 (software_nodes)
KERNEL[314.648672] remove   /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/xhci-hcd.2.auto (platform)
```
here is the event we were waiting for
```
KERNEL[314.649158] add      /devices/pci0000:00/0000:00:11.0/dwc3.0.auto/udc/dwc3.0.auto (udc)
```
after this gadget devices appear normally

htot, 2021-04-20:
Also after some time (a few minutes) the connection drop but automatically reconnects.
The following trace shows that near the end.
[lost-connection.zip](https://github.com/andy-shev/linux/files/6346323/lost-connection.zip)

htot, 2021-04-23:
In total there have appeared to be 5 separate issues, summury here:

### Throttling 
[https://lore.kernel.org/linux-usb](https://lore.kernel.org/linux-usb/CAHp75VeERhaPGAZc0HVs4fcDKXs+THc=_LFq_iEhWAR8vvURjw@mail.gmail.com/)
Caused by LPM
    • Fixed by “PROPERTY_ENTRY_BOOL("snps,dis_enblslpm_quirk")”
    • Also fixed by “BESL baseline to 6”:

diff --git a/drivers/usb/dwc3/gadget.c b/drivers/usb/dwc3/gadget.c index 60e850a395a2..423533df8927 100644
--- a/drivers/usb/dwc3/gadget.c
+++ b/drivers/usb/dwc3/gadget.c
@@ -2895,7 +2895,7 @@ static void dwc3_gadget_config_params(struct usb_gadget *g,

* recommended BESL baseline to 1 and clamp the BESL deep to be                     * within 2 to 15.
*/
- params->besl_baseline = 1;
+ params->besl_baseline = 6;
  if (dwc->is_utmi_l1_suspend)
    params->besl_deep = clamp_t(u8, dwc->hird_threshold, 2, 15);

### Problems switching DRD
[https://lore.kernel.org/linux-usb](https://lore.kernel.org/linux-usb/96c64e6a788552371081f37f544041b7ee046ef5.1618452732.git.Thinh.Nguyen@synopsys.com/)
    • Need to quickly flip switch back and forth to switch to gadget mode
    • Fixed by “usb: dwc3: core: Do core softreset when switch mode“

### Failed plug
[https://lore.kernel.org/linux-usb](https://lore.kernel.org/linux-usb/bcefaa9ecbc3e1936858c0baa14de6612960e909.1618884221.git.Thinh.Nguyen@synopsys.com/)
* appears as "can't set config #1, error -110" on the host side
* fixed by "snps,usb2-gadget-lpm-disable" or
* fixed by "usb: dwc3: gadget: Fix START_TRANSFER link state check"

### USB port resets
    • Happens also due to LPM
    • Fixed by "usb: dwc3: pci: Enable usb2-gadget-lpm-disable for Intel Merrifield"
    • This requires patch "usb: dwc3: gadget: Check for disabled LPM quirk"
    • "snps,dis_enblslpm_quirk" doesn't work to resolve port resets
    • "snps,usb2-gadget-lpm-disable" disables LPM entirely so “BESL baseline to 6” no longer required.

### Initially switch needs flipping 2x to go to device mode
    • Extcon has status variable that needs initialization
    • Fixed by “extcon-intel-mrfkd: initialize mrfld_extcon status“

andy-shev, 2021-04-24:
So, as far as I understand we need only two patches now to fix all 5 issues, i.e. extcon + `snps,usb2-gadget-lpm-disable`. Correct?

htot, 2021-04-25:
Yes. Thanks for reviewing, sent to ml.

closed 2021-04-25 by htot