andy-shev/linux #32: Support for Galileo gen 1?

dmarkey, 2021-04-05:
Hey there.

I got this repo link from Stack Overflow, I see you're working on a kernel that works on Quark.

I have 2 Gen 1 Galileos I want to try to do something interesting with before they go in the garbage (Like running a proper linux distro).

I noticed in a commit message that Gen 2 minimum is required. Does that mean Gen 1 definitely wont work and a lost cause?

dmarkey, 2021-04-10:
So after building this kernel and trying to boot I am not having any success on my Gen 1. @andy-shev is Gen 1 a dead end?

Here is my grub config for reference.
```
title Test2
    root (hd0,0)
    kernel /bzImage root=/dev/mmcblk0p3 console=ttyS1,115200n8 earlycon=uart8250,mmio32,0x8010f000,115200n8 reboot=kbd apic=debug rw debugshell=5 debug rootdelay=5 efi_main=0x000000014a65d8f3
jmp_code32=0x000000025a67744f

```

And boot sequence:
```

    GNU GRUB  version 0.97  (604K lower / 244604K upper memory)

 +-------------------------------------------------------------------------+
 | Test2                                                                   |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 |                                                                         |
 +-------------------------------------------------------------------------+
      Use the ^ and v keys to select which entry is highlighted.
      Press enter to boot the selected OS, 'e' to edit the
      commands before booting, 'a' to modify the kernel arguments
      before booting, or 'c' for a command-line.

   The highlighted entry will be booted automatically in 1 seconds.
Found layout.conf @ 0xffcff000 len 0x00000bf3

# WARNING: this file is indirectly included in a Makefile where it
# defines Make targets and pre-requisites. As a consequence you MUST
# run "make clean" BEFORE making changes to it. Failure to do so may
# result in the make process being unable to clean files it no longer
# has references to.

[main]
size=8388608
type=global

[MFH]
version=0x1
flags=0x0
address=0xfff08000
type=mfh

[Flash Image Version]
type=mfh.version
meta=version
value=0x01010000

[ROM_OVERLAY]
address=0xfffe0000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/EDKII_BOOTROM_OVERRIDE.Fv
type=some_type

[signed-key-module]
address=0xfffd8000
item_file=config/SvpSignedKeyModule.bin
svn_index=0
type=some_type
in_capsule=no

# On a deployed system, the SVN area holds the last known secure
# version of each signed asset.
# TODO: generate this area by collecting the SVN from the assets
# themselves.
[svn-area]
address=0xfffd0000
item_file=config/SVNArea.bin
type=some_type
# A capsule upgrade must implement some smart logic to make sure the
# highest Security Version Number always wins (rollback protection)
in_capsule=no

[fixed_recovery_image]
address=0xfff90000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/EDKII_RECOVERY_IMAGE1.Fv
sign=yes
type=mfh.host_fw_stage1_signed
svn_index=2
# in_capsule=no

[NV_Storage]
address=0xfff30000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/EDKII_NVRAM.bin
type=some_type

[RMU]
address=0xfff00000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/RMU.bin
type=none_registered

[boot_stage1_image1]
address=0xffec0000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/EDKII_BOOT_STAGE1_IMAGE1.Fv
sign=yes
boot_index=0
type=mfh.host_fw_stage1_signed
svn_index=1

[boot_stage1_image2]
address=0xffe80000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/EDKII_BOOT_STAGE1_IMAGE2.Fv
sign=yes
boot_index=1
type=mfh.host_fw_stage1_signed
svn_index=1

[boot_stage_2_compact]
address=0xffd00000
item_file=../../Quark_EDKII/Build/QuarkPlatform/RELEASE_GCC/FV/FlashModules/EDKII_BOOT_STAGE2_COMPACT.Fv
sign=yes
type=mfh.host_fw_stage2_signed
svn_index=3

[Ramdisk]
address=0xffa60000
item_file=../../meta-clanton/yocto_build/tmp/deploy/images/image-spi-galileo-clanton.cpio.lzma
sign=yes
type=mfh.ramdisk_signed
svn_index=7

[LAYOUT.CONF_DUMP]
address=0xffcff000
type=mfh.build_information
meta=layout

[Kernel]
address=0xff852000
item_file=../../meta-clanton/yocto_build/tmp/deploy/images/bzImage
sign=yes
type=mfh.kernel_signed
svn_index=6

[grub.conf]
address=0xff851000
item_file=grub/grub-spi.conf
sign=yes
type=mfh.bootloader_conf_signed
svn_index=5

[grub]
address=0xff800000
item_file=../../meta-clanton/yocto_build/tmp/deploy/images/grub.efi
sign=yes
fvwrap=yes
guid=B43BD3E1-64D1-4744-9394-D0E1C4DE8C87
type=mfh.bootloader_signed
svn_index=4

[Linux-EFI, setup=0x10d6, size=0x307d80]
<tumbleweed>
```

andy-shev, 2021-04-10:
It's possible to revive it, but I had played with Gen 1 like 5 years ago or so. I remember that it has 3.5mm jack for serial connection. And I even have somewhere the board :-) Dunno if it is still working. In any case I have no much time for it to spend.

That said, if you wish to invest some time, I can help you, otherwise it's not gonna work.

For the first step I would recommend you to find original stock image that you may boot with. Then it will be easier to see what can we do about.

dmarkey, 2021-04-10:
Yeah, lets go for it. I have a hella old debian distro working based on https://sourceforge.net/p/galileodebian

I have the special serial cable.

This distro comes with 3.8.7 and 3.2.0.

I can jump on some kind of instant-messaging, or here if you wish.

andy-shev, 2021-04-28:
So, can we close this now? Or you still have questions?

dmarkey, 2021-04-28:
No, Thanks for all your help!

closed 2021-04-28 by dmarkey