andy-shev/linux #36: Assistance with installation

bachejames, 2022-01-22:
Hi there!

Thank you SO much for your tireless efforts to build this repo, it is a fantastic resource. I am attempting to install Linux onto an Intel Galileo Gen 2, but with my inexperience installing Linux I'm getting rather confused.

I have been trying to follow [this StackOverflow issue](https://stackoverflow.com/questions/64904552/how-to-install-csi-tool-on-galileo-gen2-board) but am getting confused between [these instructions](https://github.com/andy-shev/buildroot/tree/intel/board/intel/common), [these instructions](https://github.com/andy-shev/buildroot/tree/intel/board/intel/galileo) and also [this repo](https://github.com/andy-shev/linux/tree/galileo) which has [these instructions](https://github.com/andy-shev/linux/blob/galileo/Documentation/admin-guide/README.rst).

Which should I use please? I thought simply using [these instructions](https://github.com/andy-shev/buildroot/tree/intel/board/intel/galileo) would be enough, but following them doesn't seem to work as the Galileo doesn't boot into Linux and am also getting confused as in the intel/common instructions, it mentions I need to build my own kernel, which is not mentioned. 

Thank you SO much!!

andy-shev, 2022-01-27:
Let's list them in the meaningful list:
1. _Buildroot_ distribution original documentation about Intel Galileo Gen 1 and Gen 2
2. _Buildroor_ distribution documentation for Intel platforms (from my repo)
3. _Linux kernel_ in-tree documentation: Documentation/admin-guide/README.rst

So, first of all let's split _Linux kernel_ and _Linux distribution_ parts. The Buildroot is a full _Linux distribution_, which includes a full infrastracture to build it from scratch. The _Linux kernel_ is just a kernel, it lacks of any userspace tools and image creation parts. That said, the 3. is needed in order if you want to build your own kernel (see below).

Buildroot distribution has support for Intel Galileo boards. However, the original support is probably being rotten over the time and may not work. If you want to have it working, you need to talk to the Buildroot maintainers. OTOH, I have a fork of it and some scripts on top with a goal to support all Intel platforms at once.  (Yeah, I would like to unify my approach that people won't be confused by these two approaches.)

Summarize above I could say you need the following:
* Build your own kernel for Intel Quark SoC (the heart of Galileo boards) with proper configuration, i.o.w. take the https://github.com/andy-shev/linux/tree/galileo branch and follow the instructions from 3. It should be as easy as running the following commands:

      git clone https://github.com/andy-shev/linux.git
      cd linux
      git checkout galileo     
      make i386_defconfig
      make
    
* Clone my https://github.com/andy-shev/buildroot/tree/intel and use command line suggested in 2. Should be something like the following:

     git clone https://github.com/andy-shev/buildroot.git
     cd buildroot
     git checkout intel
     make intel_defconfig
     make KERNEL_SRC=~/linux BOARD_INTEL_DISK_IMAGE=y BR2_TARGET_GENERIC_GETTY_PORT=ttyS1

If everything went fine, you will see the  _~/buildroot/output/images/diskimage.bin_ file which you simply copy to the micro-SD card and that's it. The BIOS should boot from it.

Just have checked that it works:

    Welcome to Buildroot
    buildroot login: root
    # uname -a
    Linux buildroot 5.17.0-rc1-00060-gbbd972713915 #7 SMP PREEMPT Thu Jan 27 19:04:59 EET 2022 i586 GNU/Linux

Note, building a full Linux distribution may take time. On home desktops it's up to dozen of hours or more depending on the configuration.

P.S. It might be useful to check this https://github.com/mdarse/galileo-resurrection. It is for Intel Galileo Gen 1, but might be still useful for Gen 2. It uses the instructions I provided to that guy.

bachejames, 2022-02-04:
Wow!!! Thank you **SOO** much Andy for such a fantastically detailed response, I hugely, hugely appreciate it.

Over the past week I've been diligently trying this, however I have not quite succeeded yet hence why I haven't quite closed this issue.

I have been able to get the Galileo to recognize the Linux image (by renaming the image to .img and writing it to the SD card, simply copying it didn't seem to be enough). However, when the Galileo arrives at the Buildroot login prompt, it does not seem to respond to any typing and instead the cursor just freezes endlessly. 

I have also not been able to connect to it via Ethernet, as that requires I upload an Arduino IDE Sketch, which I am unable to do as curiously the Arduino IDE does not recognize the Galileo once it is loaded with the Linux image.

I am writing all this in case one or both of these issues have in fact actually been caused to me not installing Linux properly. If I find this is not the case though, I'll close this issue.

andy-shev, 2022-02-04:
1. It's interesting research about image file. Usually what you need to do with the _diskimage.bin_ file is to raw copy to the storage (you may use USB stick and boot on any x86 machine just to test). The recommended software is Rufus to deploy the image on the storage (uSD card, USB stick, etc).
2. Are you trying on display or serial? I tried on serial and BuildRoot expects the main console to be on the serial interface. The prompt is for **root** user, no password (as you may see in the my reply above).
3. Hmm... First task is to get shell prompt and see what's going on there. Ethernet should be visible via `ip link list` command. By default there is no network enabled in the BuildRoot, one need to login and manually start it.
4. Oh, Arduino IDE won't recognize BuildRoot installation anyway, it requires something to be on the board side (if I remember correctly this part). And itself it has support for some old conventions and code, I wouldn't be surprosed that it won't work even if the everything else is set up and running. (As a reference you may look into https://github.com/edison-fw/meta-intel-edison/labels/Arduino issue reports against Arduino IDE for Intel Edison. Yes, the board is quite different, but I believe that the issues mentioned there more or less the same that for Galileo)

Hope above helps to figure out the current state of affairs.

That said, the BuildRoot is rather for simple testing that board works, for real usage you probably want to have Yocto, but I believe nobody is currently maintaining the images for Galileo boards.

andy-shev, 2022-02-23:
How is it going? May I close the issue?

bachejames, 2022-02-24:
Hi Andy!! 

**Thanks so much** for following up and many apologies for the delayed reply. Been trying to avoid bothering you until I could be sure of what, if any, issues remained.

Thanks to your fantastic instructions, I managed to load Linux successfully onto Galileo and then login! The issue I was having which meant that I was stuck in the login screen was that the FTDI cable I was using was faulty!!

Back to the original issue though, I'm hoping to be able to either close this issue or reply with more details super soon now. Thanks so much!!

andy-shev, 2022-03-04:
> Totally understand if you can't comment much as one of the two repos here aren't yours, but I'd be beyond grateful if you have any guidance on how the above issues can be avoided.

Besides having no time for this right now, I'm indeed not interested even to look to old 3rd party kernels. I work with latest and greatest vanilla (or at least latest LTS, which is v5.15 as of today). Good luck with the rest!

htot, 2022-03-04:
@bachejames Afaiu the kernel from dhalperi has just 2 patches on top of linux 3.5. Have you tried to apply these 2 patches on top of 5.15?

andy-shev, 2022-03-04:
> @bachejames Afaiu the kernel from dhalperi has just 2 patches on top of linux 3.5. Have you tried to apply these 2 patches on top of 5.15?

@htot  It's impossible (and it's unknown which one was the base, I suppose the old kernel, so it' unrealistic task and utter waste of time):

    Valid merge of **3.8** CSItool and **3.19** Galileo kernel trees
    Showing 4,067 changed files with **1,319,798** additions and 17 deletions.

htot, 2022-03-05:
That's a merge. But the 2 patches touch 4 and 2 files respectively.

But if you are right and CSI Tool depends on a heavily patched very old kernel, then it's dead in the water.

bachejames, 2022-03-06:
Thanks so much for both of your thoughts on this! It looks like [the instructions](https://dhalperi.github.io/linux-80211n-csitool/installation.html) suggest the tool needs at least a slightly older version to work: `These instructions are currently expected to work on Linux operating systems that are based on an upstream Linux kernel version between 3.2 (e.g. Ubuntu 12.04) and 4.2 (e.g. Ubuntu 14.04.4).` 

So I guess I might need to find a kernel and buildroot version that also supports a version in this range, though I got a bit confused by this before as in the [original Stack Overflow question](https://stackoverflow.com/questions/64904552/how-to-install-csi-tool-on-galileo-gen2-board) it was suggested that Andy's kernel and buildroot versions would be suitable for the CSI tool.

andy-shev, 2022-03-06:
> So I guess I might need to find a kernel and buildroot version that also supports a version in this range, 

In my opinion it will be a complete waste of your time, but it's your choice at the end. Good luck!

bachejames, 2022-03-10:
Gotcha! Many thanks for your really helpful advice @andy-shev and @htot.

closed 2022-03-10 by bachejames